plusvic/simplebus2-intercom #3: PIC12F508

pcalabrese, 2022-03-29:
Hi!, 
Very helpful and complete protocol description! Congrats!!

I was wondering if the PIC12F508 can be more easily replaced with a ESP8266 or other chip easier to find / program. 

I've read about your first prototypes with 433MHz transceiver and/or ESP8266. Could you share more details about?

Still Many Thanks for your great job!

plusvic, 2022-03-30:
Hi, in theory you can replace the PIC12F508 with an ESP8266, but it would a new program for decoding the signal and probably a new circuit. The ESP8266 is a lot more power hungry than the PIC12F508, that's why I ended up using the latter.

My tests with the ESP8266 where very limited, I just tried to power it from the intercom line, but it dropped the voltage too much, so I decided to switch to the PIC12F508. In theory it could be done, as the intercom line is able to power much larger devices with video screens, which require a lot of power, but I'm not an electronics expert didn't know how to do it properly.

Gurkenflieger, 2023-01-21:
Hello, do you still have your code from the ESP testing? I think that the voltage regulater overheats and it could be possible to implement a buck converter to drive an esp.

plusvic, 2023-01-23:
I didn't wrote the code for the ESP, I dropped the idea of using ESP very early in the design phase.