plusvic/simplebus2-intercom #4: Need some explanation

Elektroarzt, 2022-06-29:
Hi plusvic,

I like to use your fabulous work in a similar project (S2 bus > ESP8266 > Node Red > Apple HomeKit). Now I have some questions, that you maybe can answer:
- the PIC is not available til end of the year in Germany, is it possible to direct replace with a 12f509?
- as I understand, the UART speaks out the codes in your table, so the ESP8266 UART will fit seamless, correct?

Best regards and thanks in advance,
Thomas

plusvic, 2022-06-30:
Hi Thomas,

The only difference between the PIC 12F508 and the 509 is that the latter has more memory, so you can use it without changes in the code.

The UART speaks out the same 18-bits messages transmitted over the bus, including message id, destination address and checksum. Each message is encoded in 3 bytes when transmitted over UART. The transmissions is done at 4800 bauds, so you must configure the ESP8266 to receive at that speed.

You can see how each part of the message is decoded from the 3 bytes received over UART in https://github.com/plusvic/simplebus2-intercom/blob/main/remote_unit/src/main.cpp#L362

You will need an equivalent code for your ESP8266.

Elektroarzt, 2022-09-01:
Would it be possible for you to generate a hex file? I installed the microchip IDE but I'm experiencing lots of problems in compiling the asm file. This would be very helpful, as I am more the hardware guy. I made a PCBA out of your schematic if you or someone else is interested.

Many thanks in advance!

plusvic, 2022-09-06:
I lost the Virtual Machine where I had all the setup for building the code and generating the hex file, it would be very time-consuming for me and don't have too much time to spend on this project anymore.

closed 2022-09-06 by plusvic