plusvic/simplebus2-intercom #5: PIC12F508 Hex file

Gurkenflieger, 2023-01-26:
Is it possible to receive the hex file of the asm programm? PIC Programming is new to me and converting this to an Hex file in order to flash it through a K150 flasher is not straight forward unfortunately.

plusvic, 2023-01-26:
Unfortunately I don't have the hex file around.

Gurkenflieger, 2023-02-03:
Thanks, i managed to power an esp32 with a polulu 5v 1A DC DC converter.

Gurkenflieger, 2023-02-03:
[image]
[image]

plusvic, 2023-02-03:
The trick is measuring the time that elapses from one pin change to the other. You need to measure for how long the pin remains without changes. When that interval is around 17ms you have detected the message preamble. If you see an interval around 3ms without pin changes that's a 0, while 6ms it's a 1.

Shorter intervals, like the ones seen during the 25khz burst, are simply ignored. Every time the pin changes you simply reset the time counter. The important thing about the protocol is the time that elapses without pin changes, that's what you need to measure, and see if the interval fits any of the expected values (17ms, 6ms or 3ms)

Gurkenflieger, 2023-02-07:
Thanks, now i can send and receive messages directly with the esp 32 but if ringing outside i only receive the Acknowledgment pulses without a proper preamble and message. So i’m detecting an outside ring if there was no message and only 3 short pulses .
I think the Intercom activates over the Video, because if i ring with the CFP button (which does not activate the video) i receive the correct corresponding message with the preamble (17 ms pausetime). 

Thank you for your previous help!

plusvic, 2023-02-07:
It looks like the signal that arrives to you when someone rings outside is too weak, and falls below the threshold voltage for the LM2903. The LM2903 is acting as a voltage comparator between a fixed voltage of 500mV and the pulses that conform the signal. If the pulses are below 500mV they won't trigger the comparator. When you use the CFP button the signal is generated by your own intercom. That signal is stronger because the source is closer, and the peak voltages are higher. The same happens when you push the "open door" button in the intercom, the signal is very strong because it's generated by the intercom itself. 

In the images you posted the signal has peaks around 1.2V, but the message code is "open door", so you probably generated that image by pushing the "open door" button in your home's intercom. If the strong signal coming from the closer intercom has ~1.2V peaks, you can expect the weaker signal coming from the intercom in the street to be way lower,
probably 1/3 of that, or 400mV. 

What you can do is replacing either R3 or R4 in the [diagram](https://user-images.githubusercontent.com/182937/119274823-27513a00-bc12-11eb-81a5-44e7b55a2baf.png
) with a potentiometer. The potentiometer will allow you to adjust the voltage at pin 2 of LM2903, in your case you need to make the voltage lower, probably around 250-300mV. You don't want go too low neither, because then your signal will be affected by noise in the line.

You can try to capture with the oscilloscope the signal generated when someone rings outside, in order to be able to measure the peaks in a the signal. Of your may find the appropriate threshold voltage by trial and error, lowering the voltage a little bit until your device is able to see the signal.

Gurkenflieger, 2023-02-08:
Thanks, that helped! I used a 6k8 R4 and i can now see the ring on my oszi and in my application. Thanks alot. I also put a high pass filter on the input of the NPN transistor to prevent a long short on the bus. I have seen this on an other implementation of the simplebus 1 protocol.

plusvic, 2023-02-08:
Great! The high pass filter looks like a great design improvement. If I understand correctly you put a capacitor in between R5 and Q1, right?

Gurkenflieger, 2023-02-08:
I put the same values as C3 and R2 from the controller output to GND. Between the capacity and resistor the transistor is connected. I think if C3 and R2 are passing through the 25khz signal the values should be fine.

plusvic, 2023-02-10:
As long as you are getting a stable DC voltage at the output of the 7805 it should be fine. You can also increase the capacity of C1, that should compensate the decrease of R1.

Elektroarzt, 2023-05-11:
> I also noticed some brownouts, but the problem was the resistor value of the low pass filter. This resistor was getring pretty hot. I exchanged it for a 2.2 Ohm Resistor to minimize the voltage drop and heat buildup. With the 510 Ohm resistor and 3W power consumption of the esp, the voltage drop was massive. But the filtering is now a little worse. But still the values should filter all signals above 300Hz. What do you think ?

Gurkenflieger, do you have the described solution up and running for outside / inside ringing in your setup with ESP32?

Gurkenflieger, 2023-05-13:
Yes but i put  diode in instead of an resistor and im using 3.3V for the more efficient supply. The ESP 32 works and has no brownouts but the reference voltage of the comparator drops and therefore i cant detect a outside ring. I have to get a better scope to fix this. I also have some PCBS made for this application which fit under the intercom in the 68mm hole.
My Code works perfectly if i supply the ESP32 Externally. But at the moment my time is limited and i will rework on this in the near future.
[image]

Gurkenflieger, 2023-05-16:
> Gurkenflieger, do you have the described solution up and running for outside / inside ringing in your setup with ESP32?

So after my initial feedback i just bought a new oscilloscope with way more memory and im able to debug the system in the coming weeks. Will post my results and my schematics.

Elektroarzt, 2023-05-16:
Your board looks awesome, nice idea to put it in the junction box. Would be happy to hear from your findings :)

issalig, 2023-10-04:
> So after my initial feedback i just bought a new oscilloscope with way more memory and im able to debug the system in the coming weeks. Will post my results and my schematics.

@Gurkenflieger , is it possible for you to share your code and pcb? I want to do the same and it would save me a lot of time. No matter if I need to power it externally.