plusvic/simplebus2-intercom #6: Audio/Video capture and emission

FilippoPiazza, 2023-01-26:
Hello, very nice project.
Do you by chance have figured out how the audio is managed on the bus? What about the video?

plusvic, 2023-01-26:
Audio/Video is transmitted as an analog signal, specifically some kind of [composite video](https://en.wikipedia.org/wiki/Composite_video). I'm not an expert, so I don't really understand the details.

FilippoPiazza, 2023-01-26:
This is extremely helpful. Do you have any kind of documentation or is this your own conclusion?

plusvic, 2023-01-26:
It's my own conclusion based on some signals I was able to capture with an oscilloscope. The signals looked like this https://hackaday.com/2018/01/18/know-your-video-waveform/. So, I'm pretty sure its some sort of composite video. I also researched a little bit and learned that there are some ICs specifically designed for decoding these signals ([example](https://www.analog.com/en/products/max9526.html#product-overview)), but that's beyond my expertise level.

FilippoPiazza, 2023-01-30:
Thanks a lot

closed 2023-01-30 by FilippoPiazza