plusvic/simplebus2-intercom #7: Audio/Video protocol

gizahNL, 2023-05-24:
Do you have a description of the A/V protocol that is used on simplebus2? Would be really cool to hack together a device that can hook into the A/V stream & relay it onto my network.

plusvic, 2023-05-25:
The A/V signal is analog, not digital. It looks like some kind of [composite video](https://en.wikipedia.org/wiki/Composite_video), but I'm not sure whether its PAL, NTSC or SECAM. My expertise in that area is close to null. I had the idea of decoding the A/V signal too, but after some investigation I concluded that my knowledge about analog electronics is too limited for such undertake. If anyone with more experience can make some progress on that it would be great.

plusvic, 2023-05-25:
See #6.

issalig, 2023-10-02:
I was able to see the image with just a cheap USB video capturer. You only need gnd and as far as i remember the composite signal is the yellow cable that goes to the screen.

plusvic, 2023-10-03:
@issalig that's very interesting, can you share more details about your setup?

issalig, 2023-10-03:
I have re-opened the intercom and yes, it is the yellow cable that goes to the CVBS in the pcb silkscreen.
Upper left connector
[image]

morganflint, 2024-03-29:
Hello, great project!
I understand capturing and streaming the video with cheap devices like ESP32 might be nearly impossible but, what about just audio?
It must be possible to do it directly from the simplebus2 signal, as the Ring device does (bilateral)... but maybe the associated hardware isn't simple

Elektroarzt, 2024-03-29:
Hi ... for the basic communication part of the S2 bus except audio / video we have a solution, but audio would be very interesting. Here is a [link to an issue of a spinoff project](https://github.com/Elektroarzt/simplebus2-mqtt-bridge/issues/3), where sashahilton00 documented his investigations concerning audio so far. Maybe it helps or is a start.