plusvic/simplebus2-intercom #11: Opening a secondary entrance door

m4rcu5, 2024-11-29:
On my Comelit intercom I can control multiple doors. Without the screen turned on, hotting the "open" button wil open the primary entrance. However if I cycle through the camera feeds by pressing "1", I can open the secondary entrance when its in view by pressing the same "open" button.
Does anyone have experience with the codes this spits out?

I haven't build a module yet, as I was hoping for some experiences first, but otherwise I'll just go build, play, and report back :)

plusvic, 2024-11-29:
In my case it's very similar, but only with two doors. The codes for opening each door is the same regardless of the door you are opening. The intercom in your house simply sends the "open door" message through the wire, and the door intercom that is active at that moment is the one that actuates and opens the door asigned to it. The rest of the doors receive the message too, but they don't open their door because they are not active at that moment.

m4rcu5, 2024-11-30:
That interesting! In my case, I have a Comelit 6701W, where button "2" (sorry mistyped in the original message) is set to "self activation". If you press it, it will activate the camera/mic of the main entrance. And when pressed again cycles to the next door.

So I suspect that this "activates" the door, which will then listen to the "open" call. As far as I understand the system, the video signal is analog, and therefore allows only a single intercom to be active at the same time.

The use-case of responding to a call by opening the door would work out of the box. But my use-case would be more the other way around, opening the door from my phone, as a way of key-less entry.

Taken that the secondary door was added at a later stage, and did not require reconfiguration of my unit, I don't see the indoor unit send a addressed call to it directly. Meaning the button would send some sort of "cycle" code?
I did notice, it will always start with the first door when pressing the "self-activate", and just continue to cycle when pressed.

Is there a way to monitor for the raw codes send on the line?

plusvic, 2025-02-01:
@m4rcu5 I just discovered the answer to your question by accident. The secondary door is activated my sending the "camera on" (code 20) message twice in a short interval (around 1s). I guess that if you have a third camera it could be activated by sending the message 3 times in short intervals.

m4rcu5, 2025-02-05:
> [@m4rcu5](https://github.com/m4rcu5) I just discovered the answer to your question by accident. The secondary door is activated my sending the "camera on" (code 20) message twice in a short interval [...]

Thanks for testing! I take it you need send the code 20 twice, followed by the 'door open' code. I'll start gathering some hardware and build the circuit to test :)